x-heep/x-heep #20: add PLIC

davideschiavone, 2022-06-06:


benoitdenkinger, 2022-06-17:
I'll check this one

benoitdenkinger, 2022-06-22:
What should I connect to the PLIC for now? The uart interrupts? Should I use the external memcopy as an example to generate an interrupt? So if someone wants to add its own peripheral, there is one example.

davideschiavone, 2022-06-23:
Hi @benoitdenkinger - adding an external-internal bus like you did with masters is definitively a good idea.

In addition, we want to connect to PLIC the peripherals as in Opentitan.

CLINT RISC-V defines that some of the interrupts go directly into the core (e.g. software, timer), and the others go to a PLIC, which gets N events, and forwards 1 interrupt (called external) to the CPU - see the specs and also here: https://github.com/lowRISC/opentitan/blob/master/hw/top_earlgrey/rtl/autogen/top_earlgrey.sv#L2134

benoitdenkinger, 2022-06-23:
Ok, but now all the peripherals are not integrated (only uart). Do we let the responsibility to everyone integrating a peripheral with interrupts to connect it to the plic or should I wait for more peripheral to be integrated (SPI, DMA, ...)?

davideschiavone, 2022-06-23:
1st option, so connect only UART as for now

benoitdenkinger, 2022-06-23:
At the software level, does the interrupt handler check the interrupt source or the handler only notify there is an interrupt pending (i.e., should I read the plic registers inside the interrupt handler or outside to minimize the interrupt time)?

davideschiavone, 2022-06-24:
As you can see from here: https://github.com/lowRISC/opentitan/blob/master/hw/top_earlgrey/rtl/autogen/top_earlgrey.sv#L2134

the PLIC only produces one interrupt, which goes to the external interrupt of the CPU, thus, in ISR does not have any knowledge about which of the PLIC sources generated the interrupt. So yes, you need to read the registers of the PLIC

benoitdenkinger, 2022-06-24:
Yes I understood that, my question was not clear enough. The question is: do I read the register inside the ISR? Or do a set a flag to 1 in the ISR (so the ISR is very fast) and then in my application I first check the flag and then read the registers if needed?

davideschiavone, 2022-07-01:
done in #39

closed 2022-07-01 by davideschiavone