x-heep/x-heep #88: implement I2S and PDM peripherals

JoseCalero, 2022-08-17:
Amongst the vast amount of modalities when dealing with wearables, audio and speech are essential. In fact, they are not only used in wearables but in many other known day-to-day devices/systems (Alexa, Google Nest, etc.). Within this context, being ultra-low-power and low-cost, x-heep does not implement any I2S and/or PDM peripherals yet. Note that these are the main interfaces when dealing with microphones. Thus, designing and implementing those within x-heep is desired. Moreover, they will be validated by means of behavioral simulation and tested using a pre-trained neural network for keyword spotting speech recognition.

davideschiavone, 2022-11-04:
* PDM in progress

davideschiavone, 2023-06-16:
PDM filter has still bugs but i2s is now implemented #203

closed 2023-06-16 by davideschiavone