x-heep/x-heep #95: dma interrupt connected to fast interrupt ? e.g. #16

davideschiavone, 2022-08-20:


davideschiavone, 2022-08-20:
@benoitdenkinger @simone-machetti @ruben-roalvarez

benoitdenkinger, 2022-08-25:
I don't know how much faster it would be, but it makes sense if the platform relies a lot on the DMA. I can implement it.

simone-machetti, 2022-09-08:
Solved.

simone-machetti, 2022-09-08:
Interrupt assigned as follow:

Always-on peripherals:

-     soc_cntr: no interrupt
-     boot_rom: no interrupt
-     spi_subsystem: irq_fast[4]
-     power_manager: no interrupt
-     rv_timer_0: irq_timer
-     rv_timer_1: irq_fast[0]
-     dma: irq_fast[3]

On-off peripherals:

-     rv_plic: irq_external
-     uart: plic[8:1]
-     gpio: plic[32:9] - irq_fast[12:5]
-     i2c: plic[48:33]
-     rv_timer_0: irq_fast[1]
-     rv_timer_1: irq_fast[2]

closed 2022-10-13 by davideschiavone