x-heep/x-heep #204: FreeRTOS

fimtrey, 2023-02-22:
In the MAKEFILE it's specified to use the `LINKER` option to use the FreeRTOS library 
```make
# Linker options are 'on_chip' (default),'flash_load','flash_exec','freertos'
LINKER   ?= on_chip
```
But `LINKER=freertos` doesn't work.
> Linker specification is not correct

As fare as I can see an option like `USE_FREERTOS` could replace the following `if`s in the `sw/CMakeLists.txt` to use FreeRTOS with other applications.

```cmake
if(${PROJECT} MATCHES "freertos")
   # ...
endif()
```

davideschiavone, 2023-02-22:
assigned to @JoseCalero

davideschiavone, 2023-02-22:
can we pls delete it from the SW and Makefile ?

JoseCalero, 2023-02-22:
Yes, this is being committed to my PR for this week

fimtrey, 2023-02-23:
What command can I run to link FreeRTOS?

JoseCalero, 2023-02-23:
you just need to rename your application folder as `i2s_freertos`

fimtrey, 2023-02-23:
makes sense

closed 2023-02-23 by fimtrey