x-heep/x-heep #236: fail to compile for Verilator

gaochangw, 2023-03-29:
Hi,

I got errors when executing: 'make verilator-sim'. I wonder if it is me missing any prerequisite or it's a common issue. I compiled riscv-gnu-toolchain, Verilator and verible without problems on Ubuntu 22.04.

Error message:
ERROR: Vtestharness.mk:87: warning: overriding recipe for target 'tb_top.o'
Vtestharness.mk:85: warning: ignoring old recipe for target 'tb_top.o'
/usr/bin/ld: cannot find -lelf: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [Vtestharness.mk:91: Vtestharness] Error 1
make[1]: *** [Makefile:13: Vtestharness] Error 2

ERROR: Failed to build openhwgroup.org:systems:core-v-mini-mcu:0 : '['make']' exited with an error: 2

Many thanks!

davideschiavone, 2023-03-29:
hello @gaochangw ,

as written in the README, 

`
It has been tested only on Ubuntu 20, and we know it does NOT WORK on Ubuntu 22.
`

gaochangw, 2023-03-29:
Hi @davideschiavone, thanks for your prompt reply. My bad. I should have read it more carefully.

closed 2023-03-29 by gaochangw

davideschiavone, 2023-03-29:
no worries, also please install all the operating system requirements