x-heep/x-heep #242: Cannot transmit more than 64 words through SPI at once.

JuanSapriza, 2023-04-03:
### Triggering the problem
* Set the DMA to copy 66 words (32-bit each) **to** the Flash through SPI. 
* The process seems to finish without problems. 
* Set the DMA to copy 66 words **from** the Flash through SPI (same address as the previous words were requested to be written)
* Compare the copied data with the original data. 

### The effect of the problem
* The copied information is correct **but** for the first two bytes, which contain the content of the last two words of the original buffer, and the last two words, which contain 0xffffffff. 

### A theory
* The DMA start writing on the SPI Rx FIFO buffer correctly. 
* Once the FIFO is full, there is no proper communication between the SPI and the DMA. 
* The DMA keeps writing into the SPI FIFO, overriding the first words.  
* When the DMA then reads from the flash, the first 64 words begin with the last two words of the original buffer (which overrode the first two). 
* When the DMA tries to read the 64th and 65th words, the flash has nothing written there, so the DMA reads 0xffffffff. 

### Can be replicated in [this branch](https://github.com/JuanSapriza/x-heep/tree/spi_fifo).
It is a branch from this repository's _main_ branch with the following changes: 
* The copied data by the DMA was increased from 64 to 66 (to go beyond the limit of the SPI Rx buffer). 
* The copied data was modified to not be random, but an ordered array (initialized inside the _main_ function). 
* All read words are now printed, and in decimal, so the error can be spotted easily. 

### Process to replicate it: 
```
$ make mcu-gen
$ make app-clean
$ make app PROJECT=spi_flash_write LINKER=flash_load TARGET=pynq-z2
$ ( cd sw/vendor/yosyshq_icestorm/iceprog && make clean && make all)
$ make flash-prog
$ picocom -b 115200 -r -l --imap lfcrlf /dev/ttyUSB2 
```

This will run the project and perform the process explained at the top of this issue. It prints the comparison between the original buffer and the copied buffer at the end of the process. Ideally, what is found at both sides of the word _vs_ should be equal, meaning that the original information (left) was properly copied to the destination (right). 

### The obtained output
```
Waiting for the SPI interrupt...
triggered!
264 Bytes written in Flash at @0x00008500 
Checking write...
Waiting for the DMA interrupt...
triggered!
flash vs ram...
@00004100-@00004208 : 00 vs. 64
@00004104-@0000420c : 01 vs. 65
@00004108-@00004210 : 02 vs. 02
@0000410c-@00004214 : 03 vs. 03
. . .
@000041f8-@00004300 : 62 vs. 62
@000041fc-@00004304 : 63 vs. 63
@00004200-@00004308 : 64 vs. -1
@00004204-@0000430c : 65 vs. -1
failure, 4 errors! (Out of 66)
```
_(Some output has been cropped out for clarity). Also note that -1 stands for 0xfffffffff in the printf's decimal format._

### The expected output
```
Waiting for the SPI interrupt...
triggered!
264 Bytes written in Flash at @0x00008500 
Checking write...
Waiting for the DMA interrupt...
triggered!
flash vs ram...
@00004100-@00004208 : 00 vs. 00
@00004104-@0000420c : 01 vs. 01
@00004108-@00004210 : 02 vs. 02
@0000410c-@00004214 : 03 vs. 03
. . .
@000041f8-@00004300 : 62 vs. 62
@000041fc-@00004304 : 63 vs. 63
@00004200-@00004308 : 64 vs. 64
@00004204-@0000430c : 65 vs. 65
success! (Words checked: 66)
```

davideschiavone, 2023-06-02:
Very strange, cause the DMA has the SPI ready/valid signal exposed exactly to avoid what you suppose happens - but it definitively needs investigation - I will flag it as a bug

davideschiavone, 2024-01-11:
#416

closed 2024-01-11 by davideschiavone