x-heep/x-heep #275: Fix usages of SPI in the example apps

JuanSapriza, 2023-05-24:
Some applications that rely on the SPI do not work. 

Several defines that are used on these apps need to be corrected based on the compilation target.

* FPGA → SPI_FLASH 
* LINUX FEMU → SPI2
* Never SPI
<details>
  <summary>Example</summary>

```
#ifdef PYNQ-Z2
#USE_SPI_FLASH
#endif

int main(){
    ....
#ifdef USE_SPI_FLASH
    mmio_blablabla(...., SPI_FLASH_ADDRESS);
#else
    mmio_blablabla(...., SPI_HOST_ADDRESS);
#endif
    ...
    return 0;
}
```

</details>

Furthermore, some applications have been detected to not work under certain circumstances (using the proper definitions)

| **Application**                   | **Target** | **Linker** | **Result**                           |
|-----------------------------------|------------|------------|--------------------------------------|
| `spi_host_example`                | FPGA       | flash_exec | Stuck at the beginning               |
| `spi_host_example`                | FPGA       | flash_load | Works ok                             |
| `spi_host_example`                | Verilator  |            | Works ok                             |
| `spi_flash_write`                 | FPGA       | flash_exec | Flash writing never stops being busy |
| `spi_host_dma_power_gate_example` | FPGA       | flash_load | Works ok                             |

JuanSapriza, 2023-05-24:
Having done the following:
 
    1. Add the `ifdef`s around the `USE_SPI_FLASH` to prevent the use of `SPI_HOST` with the FPGA 
```C
#ifdef TARGET_PYNQ_Z2
#define USE_SPI_FLASH
#endif //TARGET_PYNQ_Z2
```

    2. Change `SPI2_START_ADDRESS` FOR `SPI_HOST_START_ADDRESS`

    3. Program on the FPGA with `flash_load` or `flash_exec`

The `spi_flash_write` app works properly. 
The apps `spi_host_dma_example` and `spi_host_example` manage to complete, but apparently did not write on the flash, as the read-back values are all `0xFF` (a proper read of a blank flash memory). 

Please remember to change the corresponding comments.

davideschiavone, 2023-08-11:
all are now working

closed 2023-08-11 by davideschiavone