x-heep/x-heep #280: CI to test compilation using differents configurations

JuanSapriza, 2023-05-24:
Right now, we have a CI that tests all apps using the default configurations (target: sim, linker: on_chip, etc)

We would like to have a script like the one in `X-HEEP/scripts/compile_all_apps.sh`. 

Note that this app has a bug as arrays should be defined as 
```
declare -a arr=("element1" "element2" "element3")
for i in "${arr[@]}"
do
    ....
done
```

I have tested this feature on my local computer using a variation of the script in the CI and it takes just slightly longer than the original one (the huge container configuration overhead is the real bottleneck).

davideschiavone, 2023-08-11:
@JuanSapriza , can we close this?

JuanSapriza, 2023-08-11:
If we merge #359

PR #359 (2023-08-21): Choose linker compiler and sim/nosim in test_all.sh
I added the possibility of choosing the linker, compiler and to avoid simulation in the test script. 

Source it like
```bash
. test_all.sh
```
And it will compile only with `gcc`, for `on_chip` and simulate with verilator.

Source it like 
```bash 
. test_all.sh gcc clang flash_load nosim
``` 
And it will compile for both `gcc` and `clang`, for only `flash_load` and perform no simulations. 

---

Although its not a CI, i think it will fix #280
closed 2023-08-21 by JoseCalero