x-heep/x-heep #282: Docker container for X-HEEP

JuanSapriza, 2023-05-24:
We would like to have a Docker image containing all necessary tools for X-HEEP, to make it as plug and play as possible. 

This can start from the one published on the ESL Github page for the app compilation CI.  

In it we could have Verilator, but should be careful with any proprietary files.

Guilherme-Soares-Sequeira, 2024-07-24:
Seeing as this is still opened, I figured I'd let you know we wrote a dockerfile for X-HEEP targeting the cv32e40p core and core-v toolchain, which may prove useful for this issue. It can be found [here](https://github.com/specs-feup/riscv-ai-extensions/tree/main/pulp-xheep-cv32e40p-docker).

davideschiavone, 2024-09-27:
this has been done now

closed 2024-09-27 by davideschiavone