x-heep/x-heep #310: Not able to print out floating point values

Xinhu-L, 2023-06-15:
Hi, 
I tried to print out the floating point value with the following code:
```C
int main(int argc, char *argv[])
{
    /* write something to stdout */
float __attribute__ ((noinline)) floatMul(float A, float B){
    return A*B;
}
    float floata;
    float floatb;
    float fMul;

    floata = 1.5f;
    floatb = 0.8f;
    fMul = floatMul(floata, floatb);

    printf("fMul is %f\n", fMul);

    printf("Exit");

    return EXIT_SUCCESS;
}

```

And I compile it with ARCH=rv32imfc, the uart0.log file shows these:
[image]
Why can not I print out the floating point value?

On the other hand, if I just put the floatMul function outside the main function and don't change anything else :
```
float __attribute__ ((noinline)) floatMul(float A, float B){
    return A*B;
}
int main(int argc, char *argv[])
{
...
}
```
 the compilation is OK, but the simulation will report an error:
 
[image]

Could you tell me how to solve these two problem?

davideschiavone, 2023-06-16:
I do not think you can define a function inside the main function (or any other function) AFAIK in C.

the reason you cannot print %f is most probably because the printf used (taken from the compiler) does not include the parsing of the floating format - if that is the case (look at the risc-v compiler source code to know it) you can define your own printf with float or you print hexadecimal values and then process them offline

Xinhu-L, 2023-06-17:
Thanks for your reply. Could you share a hint about how to define a func that can printf float? I have tried several ways but failed.

Xinhu-L, 2023-06-21:
Print hex format seems not work either:
``` C
#include <stdio.h>
#include <stdlib.h>
#include <math.h>

int main(int argc, char *argv[])
{
    /* write something to stdout */
    float a;
    a = 2.4;
    printf("float a:%f\n", a);
    printf("integer a:%d\n", a);
    printf("hex a:%x\n", a);

    printf("Exit");
    return EXIT_SUCCESS;
}

```
And here is the result:

[image]

davideschiavone, 2023-06-25:
you should remove the printf using the %f, also if you ask chatGPT or similar to write you a function to dump floating, it works, here is what Bing did:

`how to print a floating point number using putchar in C ?`

```

Hello, this is Bing. To print a floating point number using putchar in C, you need to convert the number into a string of digits and then print each digit using putchar. You can use the following steps:

•  Check if the number is negative. If yes, print a minus sign using putchar and make the number positive.

•  Convert the integer part of the number into a string of digits using a loop and an array. You can use the modulo operator (%) to get the last digit and the division operator (/) to remove the last digit. You can also use '0' + digit to get the ASCII code of the digit character.

•  Reverse the string of digits using another loop and an array. You can use a swap function to exchange the elements at both ends of the array until you reach the middle.

•  Print the string of digits using putchar in a loop.

•  Print a decimal point using putchar.

•  Convert the fractional part of the number into a string of digits using a loop and an array. You can use the multiplication operator (*) to get the first digit after the decimal point and then subtract it from the number. You can also use '0' + digit to get the ASCII code of the digit character.

•  Print the string of digits using putchar in a loop.

Here is an example of a function that prints a floating point number using putchar in C:

#include <stdio.h>
#include <math.h>

// A function to swap two elements
void swap(char *a, char *b)
{
char temp = *a;
*a = *b;
*b = temp;
}

// A function to print a floating point number using putchar
void putFloat(float x, int p)
{
// Check if x is negative
if (x < 0)
{
// Print a minus sign
putchar('-');
// Make x positive
x = -x;
}

// Convert the integer part of x into a string of digits
long i = (long)x; // Get the integer part
char int_str[20]; // An array to store the digits
int len = 0; // The length of the string
do
{
// Get the last digit and store it in the array
int_str[len] = '0' + i % 10;
len++;
// Remove the last digit from i
i /= 10;
} while (i > 0);

// Reverse the string of digits
for (int j = 0; j < len / 2; j++)
{
// Swap the elements at both ends
swap(&int_str[j], &int_str[len - 1 - j]);
}

// Print the string of digits
for (int j = 0; j < len; j++)
{
putchar(int_str[j]);
}

// Print a decimal point
putchar('.');

// Convert the fractional part of x into a string of digits
float f = x - (long)x; // Get the fractional part
char frac_str[20]; // An array to store the digits
len = 0; // The length of the string
while (p--)
{
// Get the first digit after the decimal point and store it in the array
f = (f - (long)f) * 10;
frac_str[len] = '0' + (long)f;
len++;
// Round up if necessary
if (fabs(f - (long)f) >= 0.5f)
{
frac_str[len - 1]++;
}
}

// Print the string of digits
for (int j = 0; j < len; j++)
{
putchar(frac_str[j]);
}
}

// A test program
int main()
{
putFloat(3.14159f, 3); // Prints 3.142
putchar('\n');
putFloat(-123.456f, 2); // Prints -123.46
putchar('\n');
}

```

davideschiavone, 2023-06-25:
I copied and pasted it and it works just fine

Xinhu-L, 2023-06-25:
I copied it and pasted in the hello_world.c, and then
```
make mcu-gen CPU=cv32e40p MEMORY_BANKS=8 
```
```
make questasim-sim FUSESOC_FLAGS="--flag=use_cv32e40p_fpu"
```
```
make app PROJECT=hello_world ARCH=rv32imfc
```

Then I start the simulation, the simulation can not exit and continue running forever... 😢 

Could you please share the details of how you tested this?

davideschiavone, 2023-06-30:
I pushed the example in the `matfadd`, it works just fine.

Xinhu-L, 2023-07-02:
Thanks for your patience, Davide.  But sorry, it still doesn't work for me. 😢 
I pulled the latest commit,
recompile the whole project with:
'''
make mcu-gen CPU=cv32e40p MEMORY_BANKS=16 FUSESOC_FLAGS="--flag=use_cv32e40p_fpu"
make questasim-sim FUSESOC_FLAGS="--flag=use_cv32e40p_fpu"
make app PROJECT=example_matfadd ARCH=rv32imfc FUSESOC_FLAGS="--flag=use_cv32e40p_fpu"
'''
and then
'''
make run PLUSARGS="c firmware=../../../sw/build/main.hex"
'''
The simulation will not exit, and got stuck, like this:
[image]

I think it might be something wrong with my computer or the environment setup, but I have solved my problem by using fix-point computing, so you can close this issue anyway.

davideschiavone, 2023-07-03:
the commands are changed, now it is easier as it uses the fusesoc parameters

```
make mcu-gen CPU=cv32e40p MEMORY_BANKS=16
make questasim-sim FUSESOC_PARAM="--FPU=1"
make app PROJECT=example_matfadd ARCH=rv32imfc
cd build/openhwgroup.org_systems_core-v-mini-mcu_0/sim-modelsim/
make run PLUSARGS="c firmware=../../../sw/build/main.hex"
```

also, I do not think that was the problem, are you sure questasim is running long enough? maybe it is just slow (given the fact that has to lead a lot of memory).

Try first with 2 numbers of banks, i.e. 

```
make mcu-gen CPU=cv32e40p
make questasim-sim FUSESOC_PARAM="--FPU=1"
make app PROJECT=example_matfadd ARCH=rv32imfc
cd build/openhwgroup.org_systems_core-v-mini-mcu_0/sim-modelsim/
make run PLUSARGS="c firmware=../../../sw/build/main.hex"
```
Otherwise try to open the waveforms and understand what is going wrong

Xinhu-L, 2023-07-03:
It works now with the new command, I simulated with the old commands for around 30 mins(should be long enough) but have no result, so I assumed that the error is due to the old commands.
Now it works, Thank you so much for your help 😄

closed 2023-07-03 by davideschiavone