x-heep/x-heep #319: create HAL for the CPU

davideschiavone, 2023-06-23:


JoseCalero, 2023-06-23:
this is in the FW planning together with other low-level HAL tasks: 
- Compiler layer abstraction.
- Handler error manager (including CPU exceptions/HardFault management)
- Board and Chip Support Packages
- etc

StMiky, 2025-02-06:
Quick reminder for future us: add a wrapper to wait for interrupt-based events, where the `mstatus.mie` bit is correctly cleared/set to avoid race conditions on critical sections (e.g., accessing the exit condition of a while loop). An example implementation could be:
```
wait_for_event(int *event_flag) {
    CSR_CLR_BIT(CSR_REG_MSTATUS, 0x8); // disable jump to ISRs
    while (*event_flag == 0) {
        wait_for_interrupt();
        CSR_SET_BIT(CSR_REG_MSTATUS, 0x8); /enable jump to ISRs
        // ISR affecting event_flag may run here
        CSR_CLR_BIT(CSR_REG_MSTATUS, 0x8); // disable jump to ISRs
    }
    CSR_SET_BIT(CSR_REG_MSTATUS, 0x8); /enable jump to ISRs
};
```