x-heep/x-heep #321: Make the software build more "silent"

JuanSapriza, 2023-06-26:
Right now errors and compilation messages get lost among all other messages that get printed. 
I think it would be useful to make the compilation "silent", printing only valuable information. In case we need to see the other outputs, we could set a flag during the making.

JuanSapriza, 2023-07-26:
change this to `OFF`
```Makefile
SET(DCMAKE_EXPORT_COMPILE_COMMANDS ON)
```

JoseCalero, 2023-07-27:
add verbose (on/off) command at the MAKE level

JuanSapriza, 2026-03-17:
I think it's silent enough. One day we should remove the fact that this message is printed for every inclusion, but that has zero priority (we would need to  change it to the makefile/cmake side instead of the .h file)  
```sh
In file included from /home/juan/Desktop/ESL/0_x-heep/sw/device/lib/sdk/timer/timer_sdk.h:21,
                 from /home/juan/Desktop/ESL/0_x-heep/sw/device/lib/sdk/timer/timer_sdk.c:12:
/home/juan/Desktop/ESL/0_x-heep/sw/device/target/sim/x-heep.h:8:9: note: '#pragma message: the x-heep.h for SIMULATION is used'
    8 | #pragma message ( "the x-heep.h for SIMULATION is used" )
      |         ^~~~~~~

```