x-heep/x-heep #332: CMakeLists.txt included source files

JuanSapriza, 2023-07-03:
As of #258 the CMakeLists.txt uses the existance of an <app>.h as parameter to determine if an app is external or not. 
External apps without such file are flagged as internal and (unless an app inside X-HEEP has the same name) will not be built. 

A patch solution is to remove
```Makefile
# Check if the APP is external to ROOT_PROJECT
if( external_app EQUAL 0 )
  SET(SOURCE_PATH ${ROOT_PROJECT})
endif()
``` 

and change `ROOT_PROJECT` for `SOURCE_PATH` in 
```Makefile
COMMAND ${CMAKE_OBJDUMP} -S ${ROOT_PROJECT}build/CMakeFiles/${MAINFILE}.elf.dir/${OBJ_PATH}applications/${PROJECT}/${SRC_MODULE}.c.obj > ${SRC_MODULE}.s
```

A proper solution should involve making the proper recognition. 

--- 

A related issue exist with the "conditional inclusion" of files:

If one of two files with the same name or that contains functions/variables with the same name is included, no conflicts should happen. 
However, because all related source files are linked, we have "redefinition problems". 
e.g.

main.c
```C
#ifdef TEST
    #include "test/functions.h"
#else
    #include "production/functions.h"
#endif
```

test/functions.h
```C
void myFunction( ) { . . .}
```

production/functions.h
```C
void myFunction( ) { . . . }
```

Will raise a re declaration error during linking.

davidmallasen, 2023-07-03:
Another patch solution is to add an empty `.h` file to any external `sw/applications/` directory which only has `.c` files.

JuanSapriza, 2023-07-26:
The problem is that the header files are compiled, then the linker includes all files, and then a second compiler pass struggles to determine which function should be used in each case. 

Consider this alternative in the declaration for the CGRA case

```C
#if CGRA_COLS == 4
    extern kcom_kernel_t strs_kernel_4X4;
    #define STRS_KERNEL &strs_kernel_4X4
#elif CGRA_COLS == 3
    extern kcom_kernel_t strs_kernel_3x3;
    #define STRS_KERNEL &strs_kernel_3x3
#endif 
```

JuanSapriza, 2023-07-27:
This is fixed in #358

closed 2023-07-27 by JuanSapriza