x-heep/x-heep #336: Include extenal C runtime files in the CMake flow

danivz, 2023-07-04:
The current CMake software compilation flow doesn't support external C runtime files.

Following the instructions in [eXtending_HEEP.md](eXtending_HEEP.md) I have:
```
BASE
    └── sw
        ├── applications
        │   └── your_app
        │       ├── main.c
        │       ├── your_app.c
        │       ├── your_app.h
        │       └── ...
        ├── build -> ../hw/vendor/esl_epfl_x_heep/sw/build
        ├── device -> ../hw/vendor/esl_epfl_x_heep/sw/device
        ├── linker -> ../hw/vendor/esl_epfl_x_heep/sw/linker
        └── external
            └── lib
                ├── crt
                |   └── external_crt0.S
                └── drivers
                │       └── fll
                |           └── fll_regs.h
                └── runtime
                    └── heepie.h
```

I want to include `external_crt0.S` in https://github.com/esl-epfl/x-heep/blob/8f2351053dd240d4e6a01ac90cfcd49fb78a747c/sw/CMakeLists.txt#L169C1-L172C37. However, If I hardcode the CRT file path in `LINKED_FILES` the compilation fails because `fll_regs.h` and `heepie.h` are included in `external_crt0.S` and the compiler can't find these headers.

I also noticed that in `crt0.S` the macro `EXTERNAL_CRT0` isn't set automatically when there is a external CRT file: https://github.com/esl-epfl/x-heep/blob/8f2351053dd240d4e6a01ac90cfcd49fb78a747c/sw/device/lib/crt/crt0.S#L44C1-L46C7

davideschiavone, 2023-07-05:
Well catched, this function should be restored asap

JoseCalero, 2023-07-27:
This should be fixed and closed due to #358

davideschiavone, 2023-08-08:
@danivz , can you please close this issue and tell us if the solution works?

JuanSapriza, 2023-08-08:
@JoseCalero you sure #358 would fix this? It did not implement what we discussed in [this comment](https://github.com/esl-epfl/x-heep/issues/336#issuecomment-1652144768)

JoseCalero, 2023-08-08:
You are right, we are missing the external, I will do it asap

JuanSapriza, 2023-08-10:
Being fixed in #364

closed 2023-09-19 by JoseCalero