x-heep/x-heep #382: example_gpio_intr not working on verilator

JuanSapriza, 2023-09-14:
In #363 we added a second interrupt in the `example_gpio_intr` application to test that the array of handlers inside the gpio hal was working. 
At that time, that was working.

Note that both interrupts are exactly the same but with different handlers. 

# Symptoms

* Run `make run-app-verilator PROJECT=example_gpio_intr` 
* Open the terminal to see the output (e.g. `screen /dev/pts/x`, replacing `x` for the screen reported in the terminal)
* The output should be
```
Write 1 to GPIO 30 and wait for interrupt...
handler 1
Write 1 to GPIO 30 and wait for interrupt...  
```
* The triggering of the second interrupt is not seen. We expect to see
```
handler 2
Success
```
at the end, but this never happens. 

# Things I have tried

* Changing the `uint8_t gpio_intr_flag` to `volatile`. → No change (but this change should be kept)
* Shortening the `PRINTF`s at the beginning of each sequence →The second `PRINTF` is not printed. 
* Not triggering the second interrupt, but instead changing the flag → The program finishes successfully. 
```C
    while(gpio_intr_flag == 0) {
        CSR_CLEAR_BITS(CSR_REG_MSTATUS, 0x8);
        gpio_intr_flag = 1;  // <---------------
        CSR_SET_BITS(CSR_REG_MSTATUS, 0x8);
    }
```

JuanSapriza, 2023-09-14:
I can tackle this. Maybe just not right now.

davideschiavone, 2023-10-04:
could this be related to https://github.com/esl-epfl/x-heep/pull/385 ?

JuanSapriza, 2023-10-04:
I doubt, but i could take a second look then.

davideschiavone, 2023-10-13:
doesnt work on questasim either as far as I can tell