x-heep/x-heep #418: Syntax Error

lpj20, 2023-10-20:
Hi,

I got some errors when executing the command "make mcu-gen". The error information as follows:

USING MINICONDA core-v-mini-mcu
/usr/bin/python util/mcu_gen.py --cfg mcu_cfg.hjson --pads_cfg pad_cfg.hjson --outdir hw/core-v-mini-mcu/include --cpu  --bus  --memorybanks  --memorybanks_il  --external_domains  --external_pads  --pkg-sv hw/core-v-mini-mcu/include/core_v_mini_mcu_pkg.sv.tpl
  File "util/mcu_gen.py", line 548
    ext_int_list = { f"EXT_INTR_{k}": v for k, v in enumerate(range(plic_used_n_interrupts, plit_n_interrupts)) }
                                   ^
SyntaxError: invalid syntax
Makefile:82: recipe for target 'mcu-gen' failed
make: *** [mcu-gen] Error 1

Many thanks!

davideschiavone, 2023-10-20:
hi, which python version are you using? did you run `make conda` and then activated it?

lpj20, 2023-11-09:
> hi, which python version are you using? did you run `make conda` and then activated it?

Got it. The question has been solved. Thx.

closed 2024-03-08 by JoseCalero