x-heep/x-heep #420: rv_timer_approximate_tick_params bug

Ho3eintaji, 2023-10-25:
I ran into a bug while using the timer in Heepocrates which probably the same story for xheep. The function "rv_timer_approximate_tick_params" which is used for setting prescale and step params based on CPU freq and desired timer freq is not working well and most of the times it returns 0 for these params

davideschiavone, 2023-10-25:
I guess it is due to the time requested and frequency at which you are running - I am not familiar with the HAL of the timer, it is from OpenTitan - maybe you want to have a look there?

davideschiavone, 2023-10-25:
changing the function should be better in my opion

Ho3eintaji, 2023-10-30:
No. atm, in my app, I check whether this happens or not:
rv_timer_approximate_tick_params(SYS_FREQ, TIMER_FREQ, &tick_params);
    if (tick_params.prescale==0){
        printf("Timer approximate function was not able to set a correct value prescale");
    }
    
Later, I will change the function inside to prevent it