x-heep/x-heep #421: example_dma_external: interrupt never triggered

grinningmosfet, 2023-10-26:
### Summary

In the `example_dma_external` example, the interrupt service routine should print `D` on the serial output via the `PRINTF` macro when an end of DMA transaction (`DMA_TRANS_END_INTR`) interrupt is triggered:

```
void dma_intr_handler_trans_done()
{
    PRINTF("D");
}
```

But it does not. The interrupt does not seem to be fired or handled.

### Steps to reproduce

Run the example_dma_external example according to the [docs](https://github.com/esl-epfl/x-heep/blob/main/docs/source/How_to/ExternalDevices.md).

- Clone the X-HEEP repository
- Open a shell in the `x-heep` directory and issue the following commands:
- `conda activate core-v-mini-mcu`
- `make mcu-gen`
- Edit `~/Desktop/x-heep/sw/applications/example_dma_external/main.c` by replacing the line `#define PRINTF_IN_SIM   0` by `#define PRINTF_IN_SIM   1`
- `make app PROJECT=example_dma_external`
- `make verilator-sim`
- `cd ./build/openhwgroup.org_systems_core-v-mini-mcu_0/sim-verilator`
- `./Vtestharness +firmware=../../../sw/build/main.hex`
- `cat uart0.log`

### Expected behaviour

Finding some `D`s into the `uart0.log` logs file.

### Actual behaviour

`uart0.log` logs the following:

```

 ===================================

  TESTING DMA ON EXTERNAL PERIPHERAL

 ===================================

 tran: 0        Ok!
 load: 0        Ok!
 laun: 0        Ok!
 >> Finished transaction.
 External DMA success
```

The program exit value is `0`. There are not a single standalone `D` printed.

JuanSapriza, 2023-11-03:
Hello! 
Firstly, thank you @grinningmosfet for the detailed issue! 

This might be related to #423 and #382. As mentioned in [this comment](https://github.com/esl-epfl/x-heep/issues/423#issuecomment-1784087792) the issue might be related to the overriding of the weak implementations. 

I will not have time to tackle this in the following weeks, but if @grinningmosfet needs it and wants to take a look at these ToDos we can try to debug it asynchronously here. 

### ToDo's
- [ ] Check if the fix of the aforementioned comment also fixes this
- [ ] Find the first commit that breaks this (to tell if the problem comes from the interrupt side or the file inclusion side)

grinningmosfet, 2023-11-20:
Hey @JuanSapriza,

While working on PR #417, I've found a way (already implemented in the DMA's HAL) to register external ISRs; see below.

Adding the following lines before the DMA initialization fixes this issue:

```
if(plic_Init()) {return EXIT_FAILURE;};
if(plic_irq_set_priority(EXT_INTR_0, 1)) {return EXIT_FAILURE;};
if(plic_irq_set_enabled(EXT_INTR_0, kPlicToggleEnabled)) {return EXIT_FAILURE;};

plic_assign_external_irq_handler(EXT_INTR_0, &handler_irq_iffifo);
```
For more context, you can check out [this commit](https://github.com/esl-epfl/x-heep/pull/417/commits/70c6e841d2a536c0a6cbc76f5595f0a7a8c5b135).