x-heep/x-heep #429: Two files needed to compile in external application

JuanSapriza, 2023-11-23:
Today @benoitdenkinger noted that if he had an external application composed of two files: 
```
BASE
├── sw
│   ├── applications
│   │   └── my_app
│   ...		├── main.c	
		└── dummy.h
```
The program compiles successfully, even if dummy is not even included. 
This does not happen if `dummy.h` is not present. 
@benoitdenkinger could you provide us the output that you get from doing so? 

I remember one of the issues was that the `SOURCE_PATH` variable was changed to `ROOT_PROJECT`.   

Might be related to #358 @JoseCalero

benoitdenkinger, 2023-11-27:
When I have only a single main.c file I get the following (this is with the heepsilon platform):

ln -sf ../hw/vendor/esl_epfl_x_heep/sw/build sw/build
Your relative sources path is ../../../../sw/
make -C hw/vendor/esl_epfl_x_heep/ app SOURCE=../../../../sw/
make[1]: Entering directory '/home/benoit/ESL/Projects/HEEPsilon/hw/vendor/esl_epfl_x_heep'
USING VENV
rm -rf sw/build
make -C sw PROJECT=cgra_size TARGET=sim LINKER=on_chip COMPILER=gcc COMPILER_PREFIX=riscv32-unknown- ARCH=rv32imc SOURCE=../../../../sw/
make[2]: Entering directory '/home/benoit/ESL/Projects/HEEPsilon/hw/vendor/esl_epfl_x_heep/sw'
$You are executing from: /home/benoit/ESL/Projects/HEEPsilon/hw/vendor/esl_epfl_x_heep/sw
$You are fetching sources from /home/benoit/ESL/Projects/HEEPsilon/sw 
if [ ! -d build ] ; then mkdir build ; fi
cd build;  \
	cmake \
	    -G "Unix Makefiles" \
		-DCMAKE_TOOLCHAIN_FILE=../cmake/riscv.cmake \
		-DROOT_PROJECT=/home/benoit/ESL/Projects/HEEPsilon/hw/vendor/esl_epfl_x_heep/sw/ \
		-DSOURCE_PATH=/home/benoit/ESL/Projects/HEEPsilon/sw/ \
		-DTARGET=sim \
		-DPROJECT:STRING=cgra_size \
		-DRISCV:STRING=/home/benoit/Tools/riscv \
		-DINC_FOLDERS:STRING=/home/benoit/ESL/Projects/HEEPsilon/hw/vendor/esl_epfl_x_heep/sw/device/target/sim/ \
		-DLINK_FOLDER:STRING=/home/benoit/ESL/Projects/HEEPsilon/hw/vendor/esl_epfl_x_heep/sw/linker \
		-DLINKER:STRING=on_chip \
		-DCOMPILER:STRING=gcc \
		-DCOMPILER_PREFIX:STRING=riscv32-unknown- \
	    ../ 
-- The ASM compiler identification is GNU
-- Found assembler: /home/benoit/Tools/riscv/bin/riscv32-unknown-elf-gcc
-- The C compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/benoit/Tools/riscv/bin/riscv32-unknown-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- [INFO] -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base/freestanding     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base/freestanding     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base/freestanding     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base/freestanding     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base/freestanding     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base/freestanding     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base/freestanding     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base/freestanding     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/base     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/ams     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/dma     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/dma     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/dma     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/fast_intr_ctrl     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/fast_intr_ctrl     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/fast_intr_ctrl     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/gpio     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/gpio     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/gpio     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/i2c     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/i2c     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/i2c     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/i2s     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/i2s     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/i2s     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/pad_control     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/pad_control     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/pdm2pcm     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/pdm2pcm     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/power_manager     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/power_manager     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/power_manager     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/rv_plic     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/rv_plic     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/rv_plic     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/rv_timer     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/rv_timer     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/rv_timer     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/soc_ctrl     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/soc_ctrl     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/soc_ctrl     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/drivers/spi_host     -I /home/benoit/ESL/Projects/HEEPsilon/sw/device/lib/d [... cut ...]
CMake Error at CMakeLists.txt:233 (add_executable):
  No SOURCES given to target: main.elf

CMake Generate step failed.  Build files cannot be regenerated correctly.
make[2]: *** [cmake/targets.mak:26: build/Makefile] Error 1
make[2]: Leaving directory '/home/benoit/ESL/Projects/HEEPsilon/hw/vendor/esl_epfl_x_heep/sw'
make[1]: *** [Makefile:118: app] Error 2
make[1]: Leaving directory '/home/benoit/ESL/Projects/HEEPsilon/hw/vendor/esl_epfl_x_heep'
make: *** [hw/vendor/esl_epfl_x_heep//external.mk:52: app] Error 2

davidmallasen, 2023-11-28:
This was an issue that happened before but with the latest commits it works fine for me (I also had to add empty header files before). Are you using an older commit of X-HEEP @benoitdenkinger ?

benoitdenkinger, 2023-11-28:
Hi @davidmallasen, the problem is now solved with the newest xheep version vendorized. Thanks!

closed 2023-11-28 by benoitdenkinger