x-heep/x-heep #470: Multiple includes of the same path

beeeeep, 2024-03-07:
Sorry if the post's format is wrong, this is the first time I am posting an issue in Github

When there are multiple header files in the same folder, that folder is included multiple times. The command LIST(REMOVE_DUPLICATES dir_list_str) in CmakeLists.txt is supposed to sort this out, but the problem is that the dir_list_str does not have any ';' to seperate each element, thus it's a single string.

A proposed fix is moving `string(REPLACE ";" "" dir_list_str ${dir_list_str})`  after adding all the paths in dir_list_str and moving `LIST(REMOVE_DUPLICATES dir_list_str)` before the previous command.

Example:

```
# Make a list of the header files that need to be included
FILE(GLOB_RECURSE new_list FOLLOW_SYMLINKS ${SOURCE_PATH}*.h)
SET(dir_list_str "")
FOREACH(file_path ${new_list})
  SET(add 0) # This variable is set to 1 if the file_pth needs to be added to the list
  if(${file_path} MATCHES "/device/")
    if(${file_path} MATCHES "/target/") # Add it if its not in target, or if its in target/${TARGET}
      if(${file_path} MATCHES ${TARGET})
        SET(add 1)
      endif()
    else()
      SET(add 1)
    endif()
  elseif(${file_path} MATCHES ${PROJECT})
    SET(add 1)
  elseif( ( ${file_path} MATCHES "/freertos/" ) AND ( ${PROJECT} MATCHES "freertos" ) )
    SET(add 1)
  elseif( ${file_path} MATCHES "/external/" )
    SET(add 1)
  endif()

# Prune list
  if( add EQUAL 1 ) # If the file path mathced one of the criterion, add it to the list
    # Get the path of the obtained directory
    GET_FILENAME_COMPONENT(dir_path ${file_path} PATH)

    # Add it to the string format list
    SET(dir_list_str ${dir_list_str} "-I ${dir_path} ")
    # Add it to the header list
    SET(h_dir_list_ ${h_dir_list_} "${dir_path}")
  endif()
ENDFOREACH()
#These two lines have been placed here to remove all duplicates from dir_list_str
LIST(REMOVE_DUPLICATES dir_list_str)
string(REPLACE ";" "" dir_list_str ${dir_list_str})
```

JuanSapriza, 2024-03-07:
Hello @beeeeep! don't dare apologize! Thanks for making the issue and properly documenting it with an explanation, code snippet and proposed solution :)

Help me understand a bit better (you can use these questions as reference for the next time also ;) ) 
1. What would the problem be? How does this situation affect what you are trying to do? You are properly explaining the cause and solution, but not what the consequence/symptom is. 
2. Could you give an example output of the original `dir_list_str` and highlighting why it's problematic? 
3. Could you give an example output of the "fixed" `dir_list_str` and highlighting why it solves the problem?

Thanks again! I'll stay alert

beeeeep, 2024-03-08:
1. There are no errors during compilation, maybe compilation takes longer but I don't have any metrics. So it's not a critical problem, it's just an overflow of compilation info when your project has a lot of files.

2. To demonstrate the problem I made an example in Hello World:
[image]

When compiling, this is a part of the generated directory list. Since the duplicates are not removed, the list contains the same directories multiple times. 
[image]

3. This is after the fix:
[image]