x-heep/x-heep #501: Unable to synthesize for Xilinx Pynq-z2

H1alus, 2024-05-04:
I want to use X-HEEP on a pynq-z2 board so I installed all the dependencies required by the wiki (verilator, risc-v gnu toolchain, verible); then I tried to synthesize the base mcu and I made the virtual env with "make venv", then sourced into it, after that I made the mcu-gen with "make mcu-gen" but the "make vivado-fpga FPGA_BOARD=pynq-z2" doesn't work and I get the error (full log in the file [make_pynq-z2.txt](https://github.com/esl-epfl/x-heep/files/15208385/make_pynq-z2.txt)):
```
ERROR: [Common 17-170] Unknown option '-freq_hz', please type 'create_bd_port -help' for usage info.

    while executing
"source {../../../hw/fpga/scripts/pynq-z2/xilinx_generate_clk_wizard.tcl}"
    (file "openhwgroup.org_systems_core-v-mini-mcu_0.tcl" line 218)
INFO: [Common 17-206] Exiting Vivado at Sat May  4 09:23:17 2024...
make[1]: *** [Makefile:8: openhwgroup.org_systems_core-v-mini-mcu_0.xpr] Error 1
make[1]: Leaving directory '/home/user/Documents/x-heep/build/openhwgroup.org_systems_core-v-mini-mcu_0/pynq-z2-vivado'
ERROR: Failed to build openhwgroup.org:systems:core-v-mini-mcu:0 : '['make']' exited with an error: 2
```

davideschiavone, 2024-05-04:
Hello @H1alus , it seems to be a vivado error, we use version 2021, which version do you have ?

H1alus, 2024-05-04:
I have installed the 2018.3 version, which pack version of Vivado 2021 are you using?

H1alus, 2024-05-08:
I confirm that downloading Vivado 2021.2 solved the issue.

closed 2024-05-08 by H1alus

shankar59, 2024-11-23:
Hi @H1alus did you run the docker based flow or did manual setup for the x-heep ?

Currently I'm facing issues with both the flows so just wanted to know what steps did you undertake apart from the ones given in the readme file.

H1alus, 2024-11-24:
I had it running under Vivado 2021.2 and Ubuntu 20.04 which should be one of the official supported OSes

shankar59, 2024-11-24:
Did you try the docker one or the manual one?
I added the settings64.sh in the ~/.bashrc but couldn't run it.

H1alus, 2024-11-24:
I was referring to the manual one. Did you restart the terminal? Is the path to vivado correct? Try running the command `vivado` in order to check if it's available, make sure to have all the other requirements for X-HEEP

shankar59, 2024-11-25:
Yes Yes. I'm a regular user of Vivado. However, I was unable to make it run with x-heep. I'll recheck my installation for x-heep and it's dependencies and try again. If you remember the commands, that were used for installing x-heep manually and setting it up, then it would be quite helpful.

H1alus, 2024-11-25:
X-HEEP is not installed but rather configured for the specific tool and setup requirements, the dependencies for the project are listed in the [docs](https://x-heep.readthedocs.io/en/latest/index.html). 

Btw,  did you create a python virtual environment inside the project root? X-HEEP relies on a specific version of FuseSOC that you should install under the virtual env. 

Make sure you meet the python requirements using the [python-requirements.txt](https://github.com/esl-epfl/x-heep/blob/main/python-requirements.txt) (assuming you are not using conda), other versions of FuseSOC don't work in my experience.

As a side note, I recommend you to get in contact with the mantainers, I was just a user for a short time but then moved on to other projects so my knowledge is limited.