x-heep/x-heep #519: Including assembly files in compilation

JuanSapriza, 2024-05-29:
```
to test the dual core configuration, I need to compile a .S file (assembly) in my application, is there a way to enhance the CMAKE flow to not only include the C files but also the ASM? it should be quite easy, I did smth similar here: https://github.com/esl-epfl/x-heep/pull/449/files
but there it was among the libraries, here I would like you to find for any ".S" file in the PROJECT folder and include it in the compilation flow - can you pls guys do it on the main heep so that I can continue with the dual core?
```
 - Davide Schiavone, circa 2024

@JuanSapriza @davideschiavone @JoseCalero

davideschiavone, 2024-05-29:
@Ho3eintaji as well

JuanSapriza, 2024-06-03:
Fixed in #522

closed 2024-06-03 by JuanSapriza