x-heep/x-heep #530: CI to check and update documentation

JuanSapriza, 2024-06-18:
I see the need for two new CIs: 
1) One that updates the Documentation on every PR merge by running 
```sh
cd docs
make clean
make html
```
Also... this commands are not working... So we need to fix that as well. 
#529 

2) One to check the type of documentation being uploaded/generated. 
I really liked the automatic generation of documentation, but the I also find it a bit hieroglyphic:
[image]

@JoseCalero @LuigiGiuffrida98 @Ho3eintaji @davideschiavone

davidmallasen, 2025-03-14:
Can we close this @JuanSapriza ?

JuanSapriza, 2025-03-14:
Good question, hadn't you solved it at some point?

closed 2025-03-17 by davidmallasen