x-heep/x-heep #545: sv2v having issue during make openroad-sky130

Wiki1998-dev, 2024-07-08:
the following error shows :
DEBUG: pre_build script 'sv2v_in_place': ['python3', 'util/sv2v_in_place.py', '--incdir-list=incdirs.txt', '--define-if=prim:SYNTHESIS', '--sv2v=sv2v', '--verbose', '--merge', 'files.txt', '-DCOCOTB_SIM=1', '-DSYNTHESIS=1', '-DVERILATOR=1'] exited with error code 1
ERROR: Failed to build openhwgroup.org:systems:core-v-mini-mcu:0 : pre_build script 'sv2v_in_place': ['python3', 'util/sv2v_in_place.py', '--incdir-list=incdirs.txt', '--define-if=prim:SYNTHESIS', '--sv2v=sv2v', '--verbose', '--merge', 'files.txt', '-DCOCOTB_SIM=1', '-DSYNTHESIS=1', '-DVERILATOR=1'] exited with error code 1

I am new in this domain ,can please guide me where is the issue?

dzemildzigal, 2024-07-30:
Hi, I'm having issues as well with the target for openroad-sky130.

It seems to be that you are experiencing the same issues as me. Here is my [more expanded ticket](https://github.com/esl-epfl/x-heep/issues/571).

dzemildzigal, 2024-07-31:
> Hi, I'm having issues as well with the target for openroad-sky130.
> 
> It seems to be that you are experiencing the same issues as me. Here is my [more expanded ticket](https://github.com/esl-epfl/x-heep/issues/571).
> 
> Long story short, yes, sv2v is having issues but it is not the root cause (I think at least). The root cause is that the sources are bugged, it's just that sv2v reports the bugs so we think sv2v is the cause, but it is just the reporter of the issue. For the past couple of days to a bit over a week I've tried changing the actual sources to push the flow through but to no success.

Upon further inspection it indeed seems to be the issue that sv2v is a bit bugged. In my linked opened issue I'm talking with other contributors that have the flow flushed out and what they suggest we do. We're waiting for more info there @Wiki1998-dev