x-heep/x-heep #566: Formatting issues on DMA.md

cousteaulecommandant, 2024-07-25:
I found the following issues in [docs/source/Peripherals/DMA.md](https://github.com/esl-epfl/x-heep/blob/main/docs/source/Peripherals/DMA.md?plain=1):
* Most lines have double spacing (or, in the case of lines that *need* the double spacing, they have quadruple spacing).
* Indentation has disappeared.
* There are some unresolved merge conflicts left in the file (`<<<<<<<` `=======` `>>>>>>>`).

The first two are specially annoying for code snippets, and may have caused the third one.

Overall, it seems that the file got mangled at some point by some code editor or a code copy-paste gone wrong.

Could someone please try to fix that file?

davideschiavone, 2024-07-30:
@JuanSapriza @TommiTerza

JuanSapriza, 2024-07-30:
Thanks @cousteaulecommandant for noticing!! Will get to it right now

PR #570 (2024-07-31): Fixed wrong formatting in dma.md
Fix  #566

@davideschiavone
closed 2024-07-31 by davideschiavone