x-heep/x-heep #571: Can't run make openroad-sky130 without failing

dzemildzigal, 2024-07-30:
While going through the X-HEEP ASIC flow I can't seem to run the openroad-sky130 target and get GDS out. 
I have OpenROAD installed globally on my machine, oss cad suite and I've ran the installations locally as well as per the documents:
https://x-heep.readthedocs.io/en/latest/How_to/ImplementASIC.html

I have all of the listed dependencies, installed globally, which is confirmed in the following:
```bash
(core-v-mini-mcu) foo@bar:~/x-heep$ which klayout
/usr/bin/klayout
(core-v-mini-mcu) foo@bar:~/x-heep$ which yosys
/usr/local/bin/yosys
(core-v-mini-mcu) foo@bar:~/x-heep$ which openroad
/usr/bin/openroad
```

And I've installed openroad locally into the x-heep/flow directory and we see that sv2v is also installed as 

```bash
(core-v-mini-mcu) foo@bar:~/x-heep$ which sv2v
/home/user/x-heep/flow/OpenROAD-flow-scripts/sv2v/bin/sv2v
```

Anyways, having activated the correct environment (using Conda), and running make openroad-sky130 we get:
```bash
.
.
.
design.sv:71104:32: Parse error: unexpected token 'and' (KW_and)
ERROR:root:Failed to run sv2v on design.sv. Exit code: 1. Full command: sv2v --exclude=assert --verbose --define=COCOTB_SIM=1 --define=SYNTHESIS=1 --define=VERILATOR=1 --incdir=../src/lowrisc_dv_dv_fcov_macros_0 --incdir=../src/lowrisc_prim_assert_0.1/rtl --incdir=../src/lowrisc_prim_util_memload_0/rtl --incdir=../src/pulp-platform.org__common_cells_1.29.0/include --incdir=../src/pulp-platform.org__register_interface_0/pulp_platform_register_interface/include --incdir=../src/openhwgroup_cve2_cve2_core_0.1/rtl ../src/lowrisc_constants_top_pkg_0/rtl/top_pkg.sv ../src/lowrisc_ip_spi_device_pkg_0.1/rtl/spi_device_reg_pkg.sv ../src/lowrisc_ip_spi_device_pkg_0.1/rtl/spi_device_pkg.sv ../src/lowrisc_prim_abstract_prim_pkg_0.1/prim_pkg.sv ../src/lowrisc_prim_cipher_pkg_0.1/rtl/prim_cipher_pkg.sv ../src/lowrisc_prim_ram_1p_pkg_0/rtl/prim_ram_1p_pkg.sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_pkg.sv ../src/lowrisc_prim_util_0.1/rtl/prim_util_pkg.sv ../src/openhwgroup.org_ip_cv32e40x_0/openhwgroup_cv32e40x/rtl/include/cv32e40x_pkg.sv ../src/openhwgroup_cve2_cve2_pkg_0.1/rtl/cve2_pkg.sv ../src/pulp-platform.org__common_cells_1.29.0/src/cb_filter_pkg.sv ../src/pulp-platform.org__common_cells_1.29.0/src/cf_math_pkg.sv ../src/pulp-platform.org__common_cells_1.29.0/src/ecc_pkg.sv ../src/pulp-platform.org__common_cells_1.29.0/src/cdc_reset_ctrlr_pkg.sv ../src/pulp-platform.org__riscv_dbg_pkg_0/pulp_platform_riscv_dbg/src/dm_pkg.sv ../src/x-heep__packages_0/addr_map_rule_pkg.sv ../src/x-heep__packages_0/obi_pkg.sv ../src/x-heep__packages_0/reg_pkg.sv ../src/x-heep__packages_0/power_manager_pkg.sv ../src/x-heep__packages_0/core_v_mini_mcu_pkg.sv ../src/x-heep_ip_power_manager_pkg_0/rtl/power_manager_reg_pkg.sv ../src/lowrisc_tlul_headers_0.1/rtl/tlul_pkg.sv ../src/pulp-platform.org__fpnew_0/pulp_platform_fpnew/src/fpnew_pkg.sv ../src/pulp-platform.org_ip_gpio_0/pulp_platform_gpio/src/gpio_reg_pkg.sv ../src/x-heep_ip_dma_0/rtl/dma_reg_pkg.sv ../src/x-heep_ip_i2s_0/rtl/i2s_reg_pkg.sv ../src/x-heep_ip_pdm2pcm_0/rtl/pdm2pcm_reg_pkg.sv ../src/x-heep_obi_spimemio_0.1.0_0/rtl/obi_spimemio_reg_pkg.sv ../src/x-heep_obi_spimemio_0.1.0_0/rtl/picorv32_pkg.sv ../src/esl_epfl_ip_cv32e40px_0/esl_epfl_cv32e40px/rtl/include/cv32e40px_apu_core_pkg.sv ../src/esl_epfl_ip_cv32e40px_0/esl_epfl_cv32e40px/rtl/include/cv32e40px_fpu_pkg.sv ../src/esl_epfl_ip_cv32e40px_0/esl_epfl_cv32e40px/rtl/include/cv32e40px_pkg.sv ../src/esl_epfl_ip_cv32e40px_0/esl_epfl_cv32e40px/rtl/include/cv32e40px_core_v_xif_pkg.sv ../src/openhwgroup.org_ip_cv32e40p_0/openhwgroup_cv32e40p/rtl/include/cv32e40p_apu_core_pkg.sv ../src/openhwgroup.org_ip_cv32e40p_0/openhwgroup_cv32e40p/rtl/include/cv32e40p_fpu_pkg.sv ../src/openhwgroup.org_ip_cv32e40p_0/openhwgroup_cv32e40p/rtl/include/cv32e40p_pkg.sv ../src/lowrisc_prim_alert_0/rtl/prim_alert_pkg.sv ../src/lowrisc_prim_all_0.1/rtl/prim_esc_pkg.sv ../src/lowrisc_ip_spi_host_1.0/rtl/spi_host_reg_pkg.sv ../src/lowrisc_ip_spi_host_1.0/rtl/spi_host_cmd_pkg.sv ../src/openhwgroup.org_ip_soc_ctrl_0/rtl/soc_ctrl_reg_pkg.sv ../src/x-heep_ip_fast_intr_ctrl_0/rtl/fast_intr_ctrl_reg_pkg.sv ../src/lowrisc_ip_i2c_0.1/rtl/i2c_reg_pkg.sv ../src/lowrisc_ip_rv_plic_example_0.1/rtl/rv_plic_reg_pkg.sv ../src/lowrisc_ip_rv_timer_0.1/rtl/rv_timer_reg_pkg.sv ../src/lowrisc_ip_uart_0.1/rtl/uart_reg_pkg.sv design.sv

DEBUG: pre_build script 'sv2v_in_place': ['python3', 'util/sv2v_in_place.py', '--incdir-list=incdirs.txt', '--define-if=prim:SYNTHESIS', '--sv2v=sv2v', '--verbose', '--merge', 'files.txt', '-DCOCOTB_SIM=1', '-DSYNTHESIS=1', '-DVERILATOR=1'] exited with error code 1
ERROR: Failed to build openhwgroup.org:systems:core-v-mini-mcu:0 : pre_build script 'sv2v_in_place': ['python3', 'util/sv2v_in_place.py', '--incdir-list=incdirs.txt', '--define-if=prim:SYNTHESIS', '--sv2v=sv2v', '--verbose', '--merge', 'files.txt', '-DCOCOTB_SIM=1', '-DSYNTHESIS=1', '-DVERILATOR=1'] exited with error code 1
path /home/ectron/dataroot/x-heep/flow/OpenROAD-flow-scripts/flow
file /home/ectron/dataroot/x-heep/build/openhwgroup.org_systems_core-v-mini-mcu_0/asic_yosys_synthesis-openroad ../src/openhwgroup.org_systems_core-v-mini-mcu_0/flow/sky130/config.mk /home/ectron/dataroot/x-heep/build/openhwgroup.org_systems_core-v-mini-mcu_0/asic_yosys_synthesis-openroad/../src/openhwgroup.org_systems_core-v-mini-mcu_0/flow/sky130/config.mk
file /home/ectron/dataroot/x-heep/build/openhwgroup.org_systems_core-v-mini-mcu_0/asic_yosys_synthesis-openroad ../src/openhwgroup.org_systems_core-v-mini-mcu_0/flow/sky130/constraint.sdc /home/ectron/dataroot/x-heep/build/openhwgroup.org_systems_core-v-mini-mcu_0/asic_yosys_synthesis-openroad/../src/openhwgroup.org_systems_core-v-mini-mcu_0/flow/sky130/constraint.sdc
file /home/ectron/dataroot/x-heep/build/openhwgroup.org_systems_core-v-mini-mcu_0/asic_yosys_synthesis-openroad ../src/openhwgroup.org_systems_core-v-mini-mcu_0/flow/OpenROAD-flow-scripts/flow/Makefile /home/ectron/dataroot/x-heep/build/openhwgroup.org_systems_core-v-mini-mcu_0/asic_yosys_synthesis-openroad/../src/openhwgroup.org_systems_core-v-mini-mcu_0/flow/OpenROAD-flow-scripts/flow/Makefile

git checkout hw/vendor/pulp_platform_common_cells/*
Updated 74 paths from the index
```

Now, starting to debug the errors out I see that the design.sv has a lot of things sv2v can't handle (things like (* async *), etc) so those things need to be manually taken out.

Later I saw, that there were connected signals to non-existing ports in various modules of the sources (we signal not existing in the obi_inst_req_t packed struct comes to mind but still being connected to obi_inst_req_t despite not existing in its port list) which fails the flow as well.

Changing ```assign core_resp_o.wpt_match = 1'b0; // Will be set by upstream wpt-module within load_store_unit``` and commenting it out completely inside of x-heep/hw/vendor/openhwgroup_cv32e40x/rtl/cv32e40x_mpu.sv because the flow fails because of it as well.

Without going into any more details, 
I'm pretty sure the target for openroad-sky130 is not flushed out at all, since I've tried multiple approaches, culminating that I have started changing the sources to push the flow through. I would like to ask for some help, and hopefully, zipped sources from the authors where this target is verifiably going through so I can compare to my state and hopefully figure this out. I think this would be the best and fastest way I could debug my issue.

If my complaint is valid, this could be a possible ticket to be opened and this issue can be fixed.

dzemildzigal, 2024-07-30:
Putting it out there that this issue is connected to issue [545](https://github.com/esl-epfl/x-heep/issues/545) opened some time ago...

dzemildzigal, 2024-07-30:
@davideschiavone I see you've originally pushed this feature, right?

davideschiavone, 2024-07-31:
The sv2v flow unfortunately is still not supported in the main branch - there is a PR that made it till the end of the flow https://github.com/esl-epfl/x-heep/pull/233 but this PR is far from being mergiable and as of today we do not have resources working on this topic - but feel free to help.

Most of the issues are into SystemVerilog interfaces and other things unsupported by SV2V.

Also, @christophmuellerorg is working on using the sv2v front end embedded into FuseSoc to target Yosys synthesis for FPGA which would also solve many issue but again, we are not working on it full time.

If you guys can help it would be nice

dzemildzigal, 2024-07-31:
I'll see if I can allocate time to get this done. I've already got some of the things in line with that PR so I'll have to revisit this.

davideschiavone, 2024-07-31:
thanks a lot - I suggest you use the FuseSoc front end for sv2v @christophmuellerorg maybe can tell you how by copying here the .core target he wrote

dzemildzigal, 2024-07-31:
Thanks @davideschiavone. I'll try the new .core as soon as it's made available by @christophmuellerorg :)