x-heep/x-heep #600: How to generate the bit-file for fpga

shankar59, 2024-11-18:
I've setup the flow using docker and I'm able to run commands like make mcu-gen and get the outputs in respective directory. However, whilst using the command make vivado-fpga FPGA_BOARD=pynq-z2 the log file shows vivado command not found. I've installed vivado in my local machine. I've added the vivado installation path to docker path and it was able to fetch the license. However i was unable to run vivado because of the error stating unsupported architecture. I'm attaching the screenshots before and after adding the vivado path to the docker for your persual.

Thanks and Regards,
Shankar.
[image]
[image]

dsjdx, 2024-12-10:
maybe you can try this commnd：
```
> sudo apt-get update
> sudo apt-get install coreutils
```
because the docker some base tools may be missing，but I don‘t know if this tone will work. If you have time，you can try Manual setup, I can generate the bit-file in my own ubuntu20.04 virtual machine.

shankar59, 2024-12-11:
For the manual flow,
Did you follow the flow as mentioned in the readme or did some additional modifications ?

Regards,
Shankaranarayanan H.

dsjdx, 2024-12-12:
In the ubuntu 18.04, 20.04 and 22.04, I both follow the flow as mentioned，you don't need to do some additional modifications, At least this is the case during the simulation stage, but I **recommend using ubuntu20.04**, because some ip(OpenTitan) in this project use the ubuntu 20.04 setting.

If you use ubuntu 20.04 , I can confirm that there are no problems with these chapters in the doc：
1: Compile with Makefile
2: DEBUG(ubuntu18.04 need to set up gcc10 manully, a little diffcult)
3: Run on FPGA
4: Simulate

There is a problem with this chapter：
Implement on ASIC：using DC flow maybe no problem(I just use the FPGA flow), but the Openroad flow has some problem(The sv2v tool will report some errors, an issue mentioned this question)

I'm still studying the other chapters, So I can’t guarantee whether there will be any problems. Hope it helps you.

shankar59, 2025-03-03:
Hi dsjdx,
               We were able to resolve the issues regarding manual flow as mentioned before. However, when we tried debugging on FPGA, we were facing some issues. Were you able to do the real-time debugging on FPGA with OpenOCD ?

Regards,
Shankar.

closed 2025-03-18 by shankar59