x-heep/x-heep #642: adding boot from spi_slave

davideschiavone, 2025-02-09:
Very similar to booting from JTAG - it was also implemented in PULPino back in the days (https://github.com/pulp-platform/pulpino/blob/master/tb/tb.sv#L251)

This will be implemented after (https://github.com/esl-epfl/x-heep/pull/617)

and will require to change both the testbench and the bootrom code and HW

JuanSapriza, 2025-02-28:
This issue is blocked by https://github.com/esl-epfl/x-heep/pull/635

---

We foresee the implementation of this like follows: 
```
HEEPidermis is at RESET state
you release the RESET
now the CPU is doing while(exit_loop_bootrom) (usually set by the testbench, never been tried on FPGA - must be tried)
Arduino loads the memories (note that the reset is released already)
Arduino writes (via SPI slave) to that exit_loop_bootrom register=1
Now the CPU exits the loop and jumps to crt0
```

The loop on the boot rom: https://github.com/esl-epfl/x-heep/blob/main/hw/ip/boot_rom/boot_rom.S#L35

As of today it is not possible to remove the SPI flash as it is in the AO peri subsystem. We can, on the other hand, remove the SPI master block.

JuanSapriza, 2025-04-11:
@davideschiavone I dont think this should have been closed