x-heep/x-heep #644: OBI req to gnt unnecessary combinatorial path

benoitdenkinger, 2025-02-11:
The memory subsystem OBI grant (gnt) signal is combinatorially depended on the OBI request (req) signal, which creates an unnecessary long combinatorial path from master->slave->master. In this specific case, the gnt signal is always high and can be fixed to 1, which I think should be the default state.

https://github.com/esl-epfl/x-heep/blob/214d22fb530c949da5cece3062ef2a7bd80ec43b/hw/core-v-mini-mcu/memory_subsystem.sv.tpl#L61

More generally, I cannot think of a reason why the gnt should be combinatorially depended on the req signal. Sequentially, yes, the request can change the next internal state of the OBI slave (e.g., to not grant the next request), but the current availability (i.e., the gnt signal) does not depend on the current request.

davideschiavone, 2025-02-24:
hi @benoitdenkinger - this comes from legacy PULP protocol - that is the father of OBI - OBI indeed removed the combinatorial path in the spec but not all the IPs have been updated - if you manage to change the BUS and all the other components and having it working it would be amazing, we can also do it together

davideschiavone, 2025-03-25:
in progress on #681

closed 2025-04-17 by davideschiavone