x-heep/x-heep #645: Trying to contact Mr. Pierre Guillod

ah-mah, 2025-02-12:
Hello,

I'm sorry for setting my problem as an issue here, but I couldn't find another way to resolve it.

I'm trying to contact Mr. Pierre Guillod for some questions about the PDM to PCM implementation,  however I always receive email delivery fail from the mail domain of EPFL.

I'd like to ask if there is any software model of the implemented architecture, since I'm working on similar project, but facing some performance issue.

Thanks a lot,
Ahmed Mahdi

JuanSapriza, 2025-02-25:
Hello Ahmed. Indeed it sees that an issue is not the best way hahaha 
Pierre has left EPFL, and we haven't had a way of contacting him. 

The proper way, though, would be to open another issue explaining in detail the problem you are facing so he or anyone else can try to help you. Once you do that we can close this issue. Alternatively, you can edit the title and description of this one. 

Regarding the software model, the PDM2PCM should be a standard block, so there are probably many models online. We do not have a software emulation of the peripheral as far as i am concerned, if that is what you mean. @JoseCalero ? 

Regarding your performance issues, maybe it is related to #272 

Regards!

JoseCalero, 2025-02-28:
Hi @ah-mah, please write an email to jose.miranda@upm.es and i will send you the models that I have

closed 2025-03-04 by davideschiavone