x-heep/x-heep #650: Unable to see Hello World output from the FPGA in the JTAG Mode

shankar59, 2025-02-24:
Hello Developers,

    We're trying to bring the x-heep on FPGA and are facing multiple issues. Could you shed some light on where we are going wrong?
Steps Undertaken:- 
1) Setting up manual flow and installation of different tools.
    We've installed Vivado, Verilator, Verible, FuseSOC in the conda environment.
    We're able to run different mcu-gen commands, then make app, make vivado and make verilator sim and get the assorted outputs in the form codes, .hex file, vivado bitstream and verilator uart log of hello world.

However, from here we are facing the issues with respect to porting the application onto the FPGA. We noticed the bootrom code is already hardcoded in the FPGA rtl itself, however, it seems like there are multiple options for dumping the application and running it like jtag, flash_load, flash_exec.

From this, we tried the jtag approach as mentioned in the Debug.md guide. We installed openOCD and got the HS-2 cable also. We tried running the openocd command, yet we didn't  see any output in the terminal. We've Pynq-z2 FPGA board. We're attaching the screenshots for your reference. 

Also, can you provide an insight on how the application gets mapped on the FPGA via this JTAG mode and is it fine, if we generate and instantiate a coe file for main.hex in the generated brams of fgpa to see whether the application is running or not ?

Thanks and Regards,
Shankar.

[image]
[image]

davideschiavone, 2025-03-04:
Hello @shankar59 - thanks for your question

I do not have an HS2 Cable on my hands now to replicate the issue but:

(1) I am not sure ( or I do not recall at the moment) that the HS2 cable has UART pins, and whether they are connected to the board. Did you check that?

(2) OpenOCD failed to connect - did you check the RESET state of X-HEEP ? you should check what is the status of the RESET buttons/switches

(3) OpenOCD alone is not enough - you would need GDB as well - please check the documentation https://x-heep.readthedocs.io/en/latest/How_to/Debug.html

shankar59, 2025-03-18:
Hi @davideschiavone, we were able to run Hello World program on FPGA both using HS2 cable and BScan option. As mentioned, Kindly update the GDB part for Debugging on FPGA too in the documentation because in the readme it is not mentioned for FPGA part. 

Thanks and Regards,
Shankar.

closed 2025-03-18 by shankar59

davideschiavone, 2025-03-18:
thanks a lot @shankar59 - can you please open a pull requests with the documentations changes you need? thx