x-heep/x-heep #661: OpenOCD + Verilator Simulation throwing Illegal Instruction

shankar59, 2025-03-04:
Hello Developers,

We tried simulating the JTAG interface using the JTAG_DPI Port, and it seems stuck at a point where it throws illegal instructions.

We tried the following commands.

     1) make mcu-gen
     2) make app

and then followed the commands as reported in the debug guide.

Kindly look into this. 

Awaiting your response.

Shankar

[image]

shankar59, 2025-03-04:
If your ~/.gdbinit is having different other program details and flags associated, then it by default enters in the python mode. 
$RISCV/bin/riscv32-unknown-elf-gdb -nx ./sw/build/main.elf by passing the -nx flag avoids this issue and you'll get the expected output.

Thanks and Regards,
Shankar.

closed 2025-03-04 by shankar59

davideschiavone, 2025-03-04:
@shankar59 - can you please make a PR updating the documentation? thx

shankar59, 2025-03-18:
Hi @davideschiavone created a PR. Please check