x-heep/x-heep #666: X-HEEP bode rate problem pynq-z2

LucaCaba, 2025-03-05:
When running the hello world program having implemented X-HEEP on a pynq z2 fpga (executing from flash with the EPFL programmer) and monitoring the output with picocom, the program only seems to be working with a baude rate of 38400 when it should work at 9600.

JuanSapriza, 2025-03-27:
@JoseCalero @LucaCaba could you please leave here the linker, compiler, compiler flags, configuration you used etc so that i can try to replicate the problem on monday? Also add the set-up you are using (how the switches and jumpers are on the pynq, if you are using an ethernet cable, etc...)

JuanSapriza, 2025-04-02:
Hello! Any updates on this? I noticed that you said you were EXECUTING FROM FLASH (my script is for loading from flash), so I tried it again with the latest version of X-HEEP and it worked just fine as well.

```sh
git pull esl main
make mcu-gen
make vivado-fpga FPGA_BOARD=pynq-z2
make vivado-fpga-pgm FPGA_BOARD=pynq-z2
make app-clean
make app PROJECT=hello_world LINKER=flash_exec TARGET=pynq-z2 COMPILER_FLAGS=-Os
make flash-prog
```

* Set both switches on the FPGA to the inside (execute from flash)
* Opened picocom with a baudrate of 9600 
```sh
picocom -b 9600 -r -l --imap lfcrlf /dev/serial/by-id/usb-FTDI_Quad_RS232-HS-if02-port0
```
* Reset the board
* Got the hello world on the terminal. 

If there is something I am doing wrong at trying to replicate please let me know.

JoseCalero, 2025-04-02:
hello @JuanSapriza , this has been solved. False alarm, something wrong was on our side.

closed 2025-04-02 by JoseCalero

JuanSapriza, 2025-04-02:
Issue #666, Lucifer's bane, cast down to eternal flames—its shadow shall haunt us no more!