x-heep/x-heep #688: [docs] SPI and SPI2 are both in the peripheral subsystem, not in AO

cousteaulecommandant, 2025-04-03:
The documentation and diagram claims that the "SPI" block is placed in the Always-on peripheral subsystem along with "SPI Flash", whereas "SPI 2" is in the Peripheral subsystem.
However, it seems that both SPI and SPI 2 are in the Peripheral subsystem with SPI Flash being the only SPI peripheral in the AO region (with two multiplexed control interfaces: spi_memio which provides memory-mapped DMA access, and spi_flash which doesn't), as shown in mcu_cfg.hjson.

Affected files:
* docs/source/index.rst
* docs/source/images/x_heep.svg
* docs/source/images/xheep_diagram.png

As a side note, while you're modifying index.rst, I would also suggest to rearrange the sections so that "Always-on peripheral subsystem domain" is introduced first, and then "Peripheral subsystem domain" after it.

Here are some suggestions for the layout of the modified diagram (depending on whether or not you decide to add the SPI MEMIO block as well, although I understand that "SPI FLASH" here refers to the whole SPI subsystem):
```
   AO PERI SUBSYS                              PERI SUBSYS
--+-------+-------+-------+-------+-------+   +-------+-------+-------+
--|   S   |   S   |   S   | S M M |   S   |   |   S   |   S   |   S   |--
  |  SOC  | POWER |  BOOT |       |       |   |       |       |       |--
--|  CTRL |  MGR  |  ROM  |  DMA  | TIMER |   |TIMER 2| PLIC  |  I2C  |
--+-------+-------+-------+-------+-------+   +-------+-------+-------+
  |   S   |   S   |   S   |S_FL  S|   S   |   |   S   |   S   |   S   |
  |F.INTR |       |       |  SPI  |  SPI  |   |       |       |       |
  | CTRL  | UART  |GPIO AO| MEMIO | FLASH |   |  SPI  | SPI 2 | GPIO  |
  +-------+-------+-------+---\_______/---+   +-------+-------+-------+
             | |      |         | | |           | | |   | | |     |
```
```
   AO PERI SUBSYS                      PERI SUBSYS
  +-------+-------+-------+-------+   +-------+-------+
--|   S   |   S   | S M M |   S   |   |   S   |   S   |--
--|       |  BOOT |       |F.INTR |   |       |       |--
  |       |  ROM  |       | CTRL  |   | PLIC  |  I2C  |
--|  SOC  +-------+  DMA  +-------+   +-------+-------+
--|  CTRL |   S   |       |   S   |   |   S   |   S   |
  |       | POWER |       |       |   |       |       |--
  |       |  MGR  |       | TIMER |   |TIMER 2| GPIO  |
  +-------+-------+-------+-------+   +-------+-------+
  |   S   |   S   |S_FL  S|   S   |   |   S   |   S   |
  |       |       |  SPI  |  SPI  |   |       |       |
  | UART  |GPIO AO| MEMIO | FLASH |   |  SPI  | SPI 2 |
  +-------+-------+---\_______/---+   +-------+-------+
     | |      |         | | |           | | |   | | |
```
```
   AO PERI SUBSYS              PERI SUBSYS
--+-------+-------+-------+   +-------+-------+
--|   S   |   S   | S M M |   |   S   |   S   |--
  |  SOC  |  BOOT |       |   |       |       |--
--|  CTRL |  ROM  |  DMA  |   | PLIC  |  I2C  |
--+-------+-------+-------+   +-------+-------+
  |   S   |   S   |   S   |   |   S   |   S   |
  |F.INTR | POWER |       |   |       |       |--
  | CTRL  |  MGR  | TIMER |   |TIMER 2| GPIO  |
  +-------+-------+-------+   +-------+-------+
  |   S   |   S   |S_F S S|   |   S   |   S   |
  |       |       |  SPI  |   |       |       |
  | UART  |GPIO AO| FLASH |   |  SPI  | SPI 2 |
  +-------+-------+-------+   +-------+-------+
     | |      |     | | |       | | |   | | |
```

davidmallasen, 2025-04-04:
Thanks for pointing it out @cousteaulecommandant. I believe this will be fixed soon with a new more detailed diagram that @JuanSapriza has made. Let's keep this issue open until we fix it to take into account your suggestions as well

davidmallasen, 2025-04-10:
Can we close @cousteaulecommandant ?

JuanSapriza, 2025-04-11:
We would still need to update the documentation.. .additionally i think we might remove the SPI 2 altogether, or at least rename it. I would leave this issue open as a reminder of updating these things as well.

cousteaulecommandant, 2025-04-22:
> Can we close [@cousteaulecommandant](https://github.com/cousteaulecommandant) ?

The diagram part no longer applies, but yes, the documentation still needs to be updated.