x-heep/x-heep #701: Clean main diagram

JuanSapriza, 2025-04-22:
Following @cousteaulecommandant suggestions in #689, I would propose we brainstorm a bit about how to clean the main diagram. I agree with him that right now its too messy and confusing. 

The idea behind having interrupts and so-on on the diagram itself is because it's far from obvious. I would really like to still have some way of showing which peripherals have a connection and to where. 

We could also add it as documentation on the PLIC/FIC. 

@davidmallasen @davideschiavone @StMiky maybe you guys have an opinion? 
@cousteaulecommandant could you add here the suggestions and pictures from the PR so they don't get lost?

JuanSapriza, 2025-04-22:
I was working on a new version of the diagram that:
* Included links to the rtl of the different blocks (we dont have proper docu for all of them yet....) -- I think here it does not work.. im curious to see if it will work on the readme preview on the repo or read the docs
* Highlighted the subsystems

[image]

cousteaulecommandant, 2025-04-28:
> [@cousteaulecommandant](https://github.com/cousteaulecommandant) could you add here the suggestions and pictures from the PR so they don't get lost?

As I mentioned in #689, the new diagram is indeed much more detailed, but it seems a bit confusing.  Here are some suggestions on what could be done to improve its readability:

* Arrows are a bit messy.  They appear to be mixing "bus interfaces" with "individual signals" (mostly interrupts), which I think should be clearly differentiated (consider using colors or dotted lines?) since they have an entirely different nature (a "32-bit interface" is not the same as 32 individual bits - it's bidirectional and includes address and control signals as well).
  In fact, I would entirely remove the "individual signals" since I think they contribute very little and make the diagram more cluttered.  (The version with all the arrows is still useful as a more complete version, but perhaps the one shown in the main README should be simplified and show only the bus interfaces.)

* I liked it more when all different subsystems were highlighted.  I see that a version with subsystems highlighted has been kept as "picture 2/2" in the diagram; maybe it should be the default.  (Once the interrupts and power manager signals have been removed, the diagram will be cleaner.)

* The background of "custom CHEEP" should be white/transparent to make it clearer that it is not part of the X-HEEP project itself (and maybe make the components semi-transparent or light-colored)

Here's a possible arrow format to differentiate all types of connection:
[image]

And an example of what the diagram looks like with only the interface arrows (much cleaner in my opinion; a version with all the arrows could still be provided as a more detailed diagram):
[image]

cousteaulecommandant, 2025-04-29:
> The idea behind having interrupts and so-on on the diagram itself is because it's far from obvious. I would really like to still have some way of showing which peripherals have a connection and to where.
> 
> We could also add it as documentation on the PLIC/FIC.

Yes; I'm not against including those somewhere (it really helps to figure out which blocks can generate interrupts and which interrupt controller they trigger), but I think that for a simple "This is X-HEEP" introductory picture (as seen on the README or the first page of the docs) it's better to have a simplified diagram with buses and blocks only.  The more detailed picture could be shown somewhere else (when talking about the interrupts, for example).  This second version could skip the subsystems since they've already been described in the other pic.

> I would keep the bus signals in black color, as they are more visible, and have the other ones in another dotted lighter color or something like that.

Fair enough; I chose green because buses were green, but it makes more sense to make them as visible as possible, and then have interrupts and other signals highlighted in blue/red/gray/etc to distinguish them.

One thing I don't like about the current diagram is how it depicts multiple signals being combined into a "signal bundle" using a "big bold bar" (which I assume is meant to be interpreted as a Simulink-style "Mux/Demux", i.e. "all these signals are combined into a bundle, and the bundle goes to this block"), but the symbol is confusing (it sort of looks like a bus and doesn't quite transmit the idea of it simply meaning "all these signals go to this module"), and it intersects other more important signals making everything more confusing.
The arrows I proposed in the previous comment were meant more as a way to say "multiple lines are grouped together into a single, thicker line" (perhaps a double line would have been better, but drawio doesn't seem to have a good option for that).

> I think that now it's already highlighted what is X-HEEP and what is the extending part in the diagram, although I'm not against making it more clear with a lighter color or transparencies.

I find the presence of the "custom CHEEP" block very confusing (especially since "CHEEP" doesn't appear to be documented anywhere in this repo), but I think the solution is to change its background to white as demonstrated above so that it clearly looks like it's "something out of X-HEEP".  Maybe changing the name to "custom X-HEEP-based SoC" helps as well.

JuanSapriza, 2025-05-02:
I agree on both that we need to change how signals are represented + CHEEP is never defined or explained. 

Regarding the clutter of the interrupts and so on... i agree it's too much and would like to have a simpler diagram... yet i would not like to have to maintain and keep in sync two diagrams :/ If there was only a way of enabling layers that would be fantastic! 

Another "feature" I would like to add is marking which peripherals are reconfigurable (i.e. can be augmented with features at mcu-gen time, or be removed, or duplicated, etc...)

JuanSapriza, 2025-05-05:
Correction needed: the external peripheral bus is connected to the AO peripheral bus