x-heep/x-heep #709: Adding Support for Vivado Simulation

shankar59, 2025-05-08:
Dear Folks,

Is there a support for vivado simulation ?

We were able to add the testbench files, however we are not sure how the memory in the x-heep is being initialized like in case of verilator we run the ./Vtestharness +firmware=../../../sw/build/main.hex.

Plus, if the support is not available, then can you shed some light on how this ./testharness+firmware command works in verilator so that we can try to map our own function with vivado ?

Thanks and Regards,
Shankaranarayanan H.

davideschiavone, 2025-05-08:
Hi @shankar59 , indeed simulation with Vivado is not supported, but if you make it work please submit a pull request

davidmallasen, 2025-05-09:
Same as #1.
I'm not sure how verilator is loading the memory. However, https://github.com/esl-epfl/x-heep/blob/f6b37e5d3cc1e9745b76161dc24978c50315ff84/tb/tb_top.cpp#L98C10-L98C20 might be a good starting point

shankar59, 2025-05-15:
Sure. We're looking into it from our side. If we are able to get it working we'll raise a pull request. Along with this, we also wanted to know about the support for post synthesis netlist simulation for ASIC ?

Thanks and Regards,
Shankaranarayanan H.